YuzukiHD/YuzukiChameleon #2: BOM generating issue for Power management IC AXP313A

hotteshen, 2022-08-31:
While ordering PCBA, the PCBA supplier told me sourcing issue of AXP1530 (U1).
I checked it in schematics and it was AXP313A, which is designed to use with H616.
But on the PCB editor, the "Device" attribute value of U1 was AXP1530.
And because of it, U! was AXP1530 instead of AXP313A in the BOM.

I quickly searched and could not find any information about AXP1530.
Is this another alternative chip? Or just some error the by package library?

Can I use AXP313A? As the supplier says they can source AXP313A.

[image]

YuzukiTsuru, 2022-08-31:
AXP1530 is the software name of AXP313A. I guess it may be named AXP1530 by internal development. After the sale, the silk screen printing is renamed AXP313A. Here, use AXP313A just right

closed 2022-08-31 by YuzukiTsuru

hotteshen, 2022-09-01:
I got it.
I already ordered PCBA with AXP313A.