YuzukiHD/YuzukiChameleon #4: easyeda can't

frayray909090, 2022-09-22:
how do i open it in easyeda

YuzukiTsuru, 2022-09-22:
maybe you need easyeda pro

hotteshen, 2022-09-22:
> how do i open it in easyeda

Go to https://pro.easyeda.com/editor and import the project files.

closed 2022-09-26 by YuzukiTsuru

frayray909090, 2022-10-23:
it still wont work
it says the file formate is wrong im putting all the stuff in so just the zip you get off of the github download button

dumtux, 2022-10-30:
If you download this GitHub repository using **Download button**, it means you zipped the whole project.
This project itself is not an EasyEDA project.
Instead, this is a meta-project including all about hardware design and documentation.
1. clone this repository to your local storage
2. navigate to `[YuzukiChameleon)/Hardware/EDAProject/`
3. you can see a zip file, import it from EasyEDA Pro (not standard)