YuzukiHD/YuzukiChameleon #5: BOM: EPHY_RTX Resistor

hotteshen, 2022-09-22:
[image]

On the schematic, the R6 is 6.04k resistor.
But it was not listed in the revised [BOM_Board1_PCB1_2022-08-28.xlsx](https://github.com/YuzukiHD/YuzukiChameleon/blob/master/Hardware/BOM%26PickAndPlace/BOM_Board1_PCB1_2022-08-28.xlsx) sheet.

R6 not present in BOM - means do not assemble?

And what does that resistor do on the `EPHY_RTX` pin?

YuzukiTsuru, 2022-09-22:
I updated the BOM, I may have accidentally deleted the information of these two resistors when I was sorting it out, sorry for that

YuzukiTsuru, 2022-09-22:
The resistance of the network PHY seems a matching resistance. In my test, it seems that the change of the resistance value will affect the network speed. It is best not to modify it.

hotteshen, 2022-09-23:
Thank you for your quick reply.
The board assembly is almost finished.
I'll test networking speed when I get the board a couple of days later.
And if I encounter a problem with networking speed, I'll add more comments below.

closed 2022-09-23 by hotteshen

YuzukiTsuru, 2022-09-23:
Note that the network interface requires an external 100M network Magnetic Modules

hotteshen, 2022-09-23:
[image]

Yes, I used PoE RJ45 jack with integrated 1:1 transformer.