YuzukiHD/YuzukiChameleon #6: PhoenixSuit Does Not Detect the Board

hotteshen, 2022-09-26:
[image]
[image]

Today the board arrived from the factory and started testing.

I powered the board on and tried to follow the Firmware flashing method in the README.
But the PhoenixSuit doesn't detect the board.
I tried several times with FEL and BOOT buttons.

I connected the board to my Linux computer and I could see it is correctly recognized as an Allwinner Device in FEL mode.
It seems like the board is working fine, but PhoenixSuit was not able to detect it.
On my Windows computer, it is recognized as just an anonymous "USB Device".

I am using Windows 10 installed on an iMac.
And I installed the Allwinner USB driver by running `install.exe` in it.
Just a few black console windows popped up and soon dissapeared.
I'm nor sure if the driver is installed correctly.

[image]

YuzukiTsuru, 2022-09-26:
* You need to install the driver manually, select usbdrv.inf and install.
* It is generally not a problem to be able to identify the USB
* If you want to use Linux to flash, you can use LiveSuit
* The test firmware supports SD card boot, just use PhoenixCard to flash into SD
You can download PhoenixCard from here: https://www.aw-ol.com/downloads?cat=5

hotteshen, 2022-09-26:
I tried to install the driver manually.
By right-clicking the USB device, and "update driver" in property dialog.
It says the driver I manually selected is already installed.
I will try to install Windows 7 and try driver installing there again.

I tried LiveShuit on my Ubuntu 22.04.
I got the following error while building AWUSB.
I also put this problem backside as I couldn't wait to see the Linux is running on this board.
So I moved to PhoenixCard.

[image]

I downloaded [tina_h616-yc_uart0.7z](https://github.com/YuzukiHD/YuzukiChameleon/releases/download/0.0.1/tina_h616-yc_uart0.7z).
I burnt this image into a Sandisk 32GB Ultra TF flash card using PhoenixCard.
After inserting it into SD slot on board, powered it on.
But the board power LED went off after 14~15 seconds.
It repeated the same way on all other boards.

When the SD card is plugged in, I could not see any USB device listed by `lsusb` command from Linux desktop.
If I unplug the SD card, the "Allwinner Technology sunxi SoC OTG connector in FEL/flashing mode" is again appeared.

YuzukiTsuru, 2022-09-26:
This GPIO is default UART TX for debug, Try to connect to a USB to UART adapt, and see the log

[image]

```
[148]BOOT0 commit : b2b1d75
[151]set pll start
[154]periph0 has been enabled
[157]set pll end
[159]unknow PMU
[160]unknow PMU
[162]PMU: AXP1530
[169]vaild para:1  select dram para0
[173]board init ok
[175]DRAM BOOT DRIVE INFO: V0.52
[178]the chip id is 0x5000
[181]chip id check OK
[186]DRAM_VCC set to 1500 mv
[188]DRAM CLK =720 MHZ
[191]DRAM Type =3 (3:DDR3,4:DDR4,7:LPDDR3,8:LPDDR4)
[197]Actual DRAM SIZE =2048 M
[200]DRAM SIZE =2048 MBytes, para1 = 310a, para2 = 8001001, dram_tpr13 = 6041
[213]DRAM simple test OK.
[215]rtc standby flag is 0x0, super standby flag is 0x0
[221]dram size =2048
[224]card no is 2
[225]sdcard 2 line count 8
[228][mmc]: mmc driver ver 2019-12-19 10:41
[232][mmc]: set f_max to 50M, set f_max_ddr to 50M
[237][mmc]: mmc 2 bias 4
[245][mmc]: ***Try MMC card 2***
[259][mmc]: MMC 5.1
[261][mmc]: HSDDR52/DDR50 8 bit
[265][mmc]: 50000000 Hz
[267][mmc]: 7456 MB
[269][mmc]: ***SD/MMC 2 init OK!!!***
[329]Loading boot-pkg Succeed(index=0).
[332]Entry_name        = u-boot
[341]Entry_name        = monitor
[345]Entry_name        = dtbo

[348]Entry_name        = dtb
[351]tunning data addr:0x4a0003e8
[355]Jump to second Boot.
```

hotteshen, 2022-09-26:
[image]

I connected a USB UART adapter and I could see the log messages.
It was a way longer error message.
Here are a few last lines of the message.

As the messages were too long, I archived all 10 screenshots tho the following attached archive.

[Screenshots of error messages.zip](https://github.com/YuzukiHD/YuzukiChameleon/files/9646497/Screenshots.of.error.messages.zip)

I will try enabling debug mode in PhoenixCard later, but I suspect it might give the same result.

hotteshen, 2022-09-26:
I compared my log messages with your successful ones.
And found that my board is stopping just after `DRAM_VCC set to 1500 mv`

[image]

YuzukiTsuru, 2022-09-26:
This error indicates that the attempted dram timing parameters failed to start, and he will try the remaining dram timing parameters. Seeing dram simple test ok means that there is no problem with the dram.

It seems that you are using the `量产卡` mode, try re-burning to use the `启动卡` mode

hotteshen, 2022-09-26:
Is there a default RX pin on the GPIO pin to interact with the system via the UART terminal?
I am now only seeing the logs, and no way to input commands to play.

YuzukiTsuru, 2022-09-26:
The blue one is RX

[image]

hotteshen, 2022-09-26:
Thanks, I am now accessing the Tina Linux terminal via UART.

[image]

closed 2022-09-26 by hotteshen

ndmgrphc, 2022-09-26:
@hotteshen can I ask what factory you used?

hotteshen, 2022-09-26:
I used a factory belonging to my company's PCBA division. I am an R&D engineer at a tech company in Shenzhen.
You can email to me for discussing PCBA.
hotteshen@gmail.com