YuzukiHD/YuzukiChameleon #7: How to run an Ubuntu OS?

hotteshen, 2022-09-26:
So far, I could run Tina Linux in the [RELEASE](https://github.com/YuzukiHD/YuzukiChameleon/releases) page.

Though it is said to support Tina Linux, Ubuntu, and Android TV firmware, there's only Tina Linux source code and image on the release page. **Is there a quick way to run Ubuntu or Android using prebuilt distro images?**

I watched [your video on Bilibili](https://www.bilibili.com/video/BV1KB4y1z79U/).
There I could see you're burning some `Orangep...epi.img` file using Balena Etcher.
On booting, it shows it's *Focal* and it went into a desktop environment.
So I tried OrangePi Zero2's Ubuntu Focal Fosa image from [their storage](https://drive.google.com/drive/folders/1ohxfoxWJ0sv8yEHbrXL1Bu2RkBhuCMup).
However, writing it into an SD card using Balena etcher doesn't boot at all.

hotteshen, 2022-09-26:
I discovered that the [MangoPi MQ Quad](https://mangopi.org/mqquad)'s Ubuntu image is compitible.
I believe it is the same one used in [your video on Bilibili](https://www.bilibili.com/video/BV1KB4y1z79U/).

As MangoPi MQ Quad has a different wireless module, I think it will not support Wifi yet.
And at this moment, that ubuntu image does not support the wireless module on MangoPi itself.

YuzukiTsuru, 2022-09-27:
I am currently preparing the driver and configuration of the mainline. Recently, the course is tight and I don't have time to do these amateur projects. The hardware of the board is basically the same as that of MangoPi, you can refer to his for modification, sorry for that

hotteshen, 2022-09-27:
Understood.

Using MangoPi's Ubuntu/Android image on Chameleon has two problems.
* Wifi/BT does not work, as each uses different wireless chipsets
* Ethernet is not working as MangoPi MQ Quad does not have an Ethernet port

Ethernet may be a quick fix without rebuilding the kernel, or not?
For the Wireless module, if you have some idea to custom-build Ubuntu (or Armbian) to use it,
please let me know. I am gonna try, but there is little documentation about H616 and XR829 on the Internet.

(My wechat id is `hotte2021` it would be great if I can query quick questions in chat, instead of opening so many issue tickets here. This line will be deleted once you reply here on wechat)

YuzukiTsuru, 2022-09-27:
Ethernet you only need to add it in the kernel, wifi driver I will publish it when I done. You can try to download Tina Linux SDK from MangoPi and enable the wifi driver and ephy driver, I tested it's working.

(I think it is better to communicate in the issue, which can help other people debug this board)

hotteshen, 2022-09-27:
That sounds great.
I will try to enable Ethernet and Wifi in that way.

dumtux, 2022-11-23:
Debian build is working now.
Check out this repository [Allwinner-H616](https://github.com/dumtux/Allwinner-H616)

closed 2022-11-23 by dumtux