YuzukiHD/YuzukiChameleon #8: License for the forked or modified Board, Can it be used for a commercial project?

hotteshen, 2022-09-27:
The License of this project is defined as **CERN Open Hardware Licence Version 2 - Strongly Reciprocal**.
I would like to know about the following cases, as I don't have a clue understanding about CERN Open Hardware Licenses.

* I would like to use this board for a commercial project. The client does not want to put the design in public. Can I?
* I would like to publish my modified version of this board, e.g. upgrade for BT5.0 on GitHub, with the same open license, but another name. Can I?

YuzukiTsuru, 2022-09-27:
A: Roughly speaking, the hardware equivalent of a binary object in the
free and open-source software world would be a tangible object made
thanks to design files licensed under an OSHW licence. So implementing a
reciprocal licence involves solving the challenge of ensuring that the
recipient of a piece of hardware gets access to the original design
files. This is tricky because copyright law (which is the mechanism used
by most software licences to impose obligations on licensees) does not
apply to hardware production and distribution. In drafting CERN-OHL-S,
we have made a special effort to achieve this effect by relying solely
on existing and applicable rights, and we believe the licence will
fulfil this part of its purpose in a large number of cases. Read on to
learn how CERN-OHL-S provides two mechanisms for a recipient of
hardware to get hold of design documents:

- The licence explicitly asks a Licensee to provide an easy way to
    access the design Source to recipients of products when (s)he makes
    and distributes products based on CERN-OHL-S licensed designs.
- For cases in which a tangible piece of hardware changes hands many
    times before reaching the final recipient, it is impossible to
    enforce the obligations above without making the licence text much
    more complicated and incurring in many extra contractual relations.
    The original designer though, can make it much more likely for the
    recipient to get access to the design files by including a URL or
    similar in the design, in a place where it will result in a visible
    imprinted label in the final object. The CERN-OHL-S requests that
    downstream Licensees respect that Notice and modify it accordingly
    if they modify the design files. It also requires that they make the
    Complete Source for the Product available to the recipient of any
    Product. The easiest way to do this, and the one which we encourage,
    is to make the Complete Source available from a Source Location
    (which is typically a public repository). However, it is also
    possible (although more cumbersome), to distribute Source and a
    Product privately, so long as the recipient is provided with a copy
    of the Complete Source.
Methods to enforce the licence will depend on each case. For example,
most of the hardware designs are likely to be stored and distributed in
the form of design files, i.e. digitally, and doing just about anything
with them is going to involve copying, at a minimum, which will require
a licence, and therefore allows the licence to impinge. Once the
licensee exercises any rights under the licence, the whole licence text
applies, including obligations at production and distribution time. This
is just one example involving copyright, but as we said in an answer to
another question, there may be more rights the licence rests upon in
each case. It is important for a licensor to understand how the licence
is intended to work and what can and cannot be achieved for a particular
design in a particular context. No licence gives licensors 100%
reassurance of what will happen in court one day, and the CERN OHL is
not an exception in that
respect.

-- From: https://ohwr.org/project/cernohl/wikis/FAQ#q-copyright-does-not-cover-hardware-how-do-you-implement-strongly-reciprocal-licensing-in-cern-ohl-s

hotteshen, 2022-09-27:
Makes sense.
So it is more like GPL, less like BSD for software licenses.
Thank you.

closed 2022-09-27 by hotteshen