YuzukiHD/YuzukiChameleon #9: How to add an Activity LED?

hotteshen, 2022-09-27:
Raspberry Pi and some other SBCs have an activity indicating LED aside from a power indicating LED.
It is somewhat similar to the HDD light of computers.
Does H616 have a dedicated pin for the activity indicator?

YuzukiTsuru, 2022-09-28:
no, you need config it in software by writing a driver

dumtux, 2022-10-30:
I could add Ethernet Activity /  Link LEDs using the device tree.

```
	leds {
		compatible = "gpio-leds";

		system_led: system {
			gpios = <&gpio1 0 GPIO_ACTIVE_LOW>;
		};
		eth0_led0: link {
			label = "omega2p:green:link";
			gpios = <&gpio 18 GPIO_ACTIVE_LOW>;
			default-state = "off";
		};
		eth0_led1: data {
			label = "omega2p:orange:data";
			gpios = <&gpio 19 GPIO_ACTIVE_LOW>;
			default-state = "off";
		};
	};
```

closed 2022-10-30 by dumtux