YuzukiHD/YuzukiChameleon #10: Ideas on changing HDMI to MIPI ports

thekidtonystark, 2022-10-15:
I plan on using three of these within a VR headset, as a sort of multiplexer for 12 cameras maximum! Four will be connected to each board but some help turning the HDMI to MIPI would be great! Can't use a converter PCB as they are way too costly!

YuzukiTsuru, 2022-10-15:
In fact, H616 is an incomplete chip version, which is designed for OTT special-purpose use. The actual complete chip is an industrial level chip called T507, The core, peripheral and register list are the same

[image]

thekidtonystark, 2022-10-15:
Thank you! Looks like I have some work to do on my end.

closed 2022-10-17 by YuzukiTsuru