YuzukiHD/YuzukiChameleon #13: Wi-Fi / Bluetooth chip XR829 mainline driver

dumtux, 2022-11-04:
I could find the [mainline driver for XR819](https://github.com/fifteenhex/xradio).
But I could not find anything for XR829 used in Chameleon.
Is there a driver for mainline kernel in public?

YuzukiTsuru, 2022-11-04:
You can try this. I haven't found the main driver of the XR829. This is a newer version of the bsp driver 
https://github.com/YuzukiTsuru/Xradio-XR829

dumtux, 2022-11-05:
I tried it.
But couldn't make it work yet.

How to properly configure FEX/dts files for our board.

dumtux, 2023-01-31:
We could not make XR829 work on the mainline kernel and Ubuntu.
We decided to update the hardware design using AW859, as we could find a mainline driver for it.

If XR829 mainline driver is not released by Allwinner, this issue is not for YuzukiChameleon project.
It would be raised on some kernel projects of Allwinner.
So, I would like to close this issue ticket.

closed 2023-01-31 by dumtux

YuzukiTsuru, 2023-05-08:
Hi, We just release a driver for linux kernel 5.15, try this driver for mainline 
https://github.com/YuzukiHD/Xradio-XR829/tree/5.15

dumtux, 2023-07-09:
Hi @YuzukiTsuru 
I added the X829 driver support to my [Allwinner-H616 armbian build](https://github.com/dumtux/Allwinner-H616).
Though the driver binary files should be hard-coped to the SD / eMMC after build and flash, it is working anyhow.