YuzukiHD/YuzukiChameleon #19: wlan0 no scan result

leonGYP, 2023-08-07:
I use the newest image which support wifi，but in my case "wlan0" exists，but no scan result with command "iwlist wlan0 scan", how can i connect wifi with the newest image, please help!

VilriginMCH, 2023-12-08:
i got the same question

YuzukiTsuru, 2023-12-08:
The XR829 not supported in rfkill, needs more config for it

VilriginMCH, 2024-01-07:
> The XR829 not supported in rfkill, needs more config for it

Thx for your answer and I'd like to know how can I test my XR829 device on board or connect to the wifi manually?

hawav, 2024-05-13:
I'm encountering the same issue with the XR829 device. Have you made any progress in resolving the rfkill support or found any additional configurations that might help? I'm interested in knowing if there are any updates or suggestions you can provide

VilriginMCH, 2024-05-21:
@hawav @leonGYP I've tackled this problem. You can create a subnode of rfkill and move your wlan0 subnode under it. Remember to modify your REGON pin, properly config the pwrseq and copy the proper firmware bin file(`mv /lib/firmware/sdd_xr829_40M.bin /lib/firmware/sdd_xr829.bin`).

在设备树中新建一个rfkill的子节点，把wlan0子节点挪到它的下面去。别忘了在wlan0子节点中设置好REGON引脚，设置好正确的sdmmc上电时序(也就是pwrseq)并将拷贝正确的固件。

closed 2025-06-21 by chainsx